rice-eclipse/rice-eclipse-mk1-1 #53: Improve the look of the GUI

rjcrjc, 2017-12-17:
The final changes to the GUI were made with no understanding of Tkinter and the GUI looks terrible. Make the GUI usable and also make the positions somewhat dependent on the size of the window. That is, the buttons should still show up even if the window is resized to be relatively small.

Ideally, we would also have tabs for different data displays.

ty19, 2017-12-22:
The GUI (pushed to final-code-c) looks nicer now, imo. The right panel is ugly because it's mostly empty, but we can always add things later. I set the window to be non-resizable, which I consider a fix to the window size issue.

[image]
 It seems like there's a built-in way to do tabs, but I'm not sure what I would put in each one. We could do something like default (important ones we look at during tests), LCs, PTs, TCs for each tab. Do you have something specific in mind?

rjcrjc, 2017-12-24:
I did not have anything specific in mind for tabs, but I have a couple things that should probably go in their own tabs: Namely a tab for any output from the loggers. I would also have a tab for each sensor that includes calibration and some data such as an average over last 10 seconds.

I suppose the window resizing fix is probably the easiest thing without a lot of tkinter work that isn't really a good use of our time.