rice-eclipse/rice-eclipse-mk1-1 #54: Fix real-time graphing.

rjcrjc, 2017-12-17:
The graphs on the GUI do not discard old data and they also display the data with all data-points. Furthermore, they do not look good with noisy data.

The graphs should have the following two properties: 
* Ability to display a specified range of time (say 10 seconds, 1 minute etc.)
    * For large times like a minute or more, the graphing code will need to downsample the data in order to avoid plotting too many points. Contact me for help with proper downsampling and decimation.
* Option for denoised plots. Again contact me or look at the analytics.m script I posted in the google drive for how to clean up the noise.

ty19, 2017-12-22:
I added a basic way to change time ranges. I guess time would just be total_samples / samples_per_time if we put the sample rate of each sensor somewhere in the code. I can add buttons and stuff so we can change the time ranges during tests. Not sure on how to downsample other than by throwing out every nth data point in the data we're plotting.  

[image]

Is there any reason denoised plots would not be the default option?

rjcrjc, 2017-12-24:
Nice work on this stuff! I can help with proper denoising and downsampling later since it's not really in the realm of CS knowledge, but for now just throwing out points is good enough for the real-time plots.